gvanem/Dump1090 #2: Map is empty, data.json empty, interactive readout shows flights

chadhutchins182, 2022-02-07:
Hello!

I have an issue where the map shows no flights, the web service is returning no data, but the interactive readout shows flights.

[image]

Any assistance on how to get the map working would be greatly appreciated.

gvanem, 2022-02-07:
How did you start the program? With what options? 
And try a `set DUMP1090_HOMEPOS=latitude,longitude` to show the distance to the plane(s).

chadhutchins182, 2022-02-07:
I was using git bash on Windows. 

Command: ` ./dump1090.exe --interactive --net`

I just tried PowerShell and cmd and I get something like this:

```
Dump1090>dump1090.exe --interactive --net
Invalid home-pos 32.691144,-114.587513
Decoder statistics:
        0 valid preambles.
        0 demodulated after phase correction.
        0 demodulated with zero errors.
        0 with CRC okay.
        0 with CRC failure.
        0 errors corrected.
        0 messages with 1 bit errors fixed.
        0 messages with 2 bit errors fixed.
        0 total usable messages.
        0 unique aircrafts.
        0 unique aircrafts from CSV.
        0 unrecognized ME types.

Network statistics:
  Raw TCP output (port 30002):
    Nothing.
  Raw TCP input (port 30001):
    Nothing.
  SBS TCP output (port 30003):
    Nothing.
  SBS TCP input (port 30003):
    Nothing.
  HTTP server (port 8080):
           0 HTTP GET requests received.
           0 HTTP/WebSocket upgrades.
           0 server connection "keep-alive".
           0 client connection "keep-alive".
    Nothing.
```

How I set home position:
`set DUMP1090_HOMEPOS=32.691144,-114.587513`

Something seems incorrect with HOMEPOS, how should it be set?

gvanem, 2022-02-07:
> `set DUMP1090_HOMEPOS=32.691144,-114.587513`

I've never tested with a negative longitude; the macro for `VALID_POS(pos)` broke when something is west of Greenwich  meridian.

This seems to be the cause of `.json`-data not being sent to the Web-client.
Fixed now in https://github.com/gvanem/Dump1090/commit/255a72c956f3e4274aa800d2e6fd77177365087e. Please try with the new `dump1090.exe`.

chadhutchins182, 2022-02-07:
Different error this time:

```
C:\Users\chad\Documents\CODE\Dump1090>dump1090.exe --interactive --net
Aircraft database "C:\Users\chad\Documents\CODE\Dump1090\aircraftDatabase.csv" does not exist.
Found 1 device(s):
0: Realtek, RTL2838UHIDIR, SN: 00000001
Error opening the RTLSDR device -1: WinUsb-error -1.
Decoder statistics:
        0 valid preambles.
        0 demodulated after phase correction.
        0 demodulated with 0 errors.
        0 with CRC okay.
        0 with CRC failure.
        0 errors corrected.
        0 messages with 1 bit errors fixed.
        0 messages with 2 bit errors fixed.
        0 total usable messages.
        0 unique aircrafts.
        0 unique aircrafts from CSV.
        0 unrecognized ME types.

Network statistics:
  Raw TCP output (port 30002):
    Nothing.
  Raw TCP input (port 30001):
    Nothing.
  SBS TCP output (port 30003):
    Nothing.
  SBS TCP input (port 30003):
    Nothing.
  HTTP server (port 8080):
           0 HTTP GET requests received.
           0 HTTP/WebSocket upgrades.
           0 server connection "keep-alive".
           0 client connection "keep-alive".
    Nothing.
```

gvanem, 2022-02-08:
Oh, my bad. Fixed in https://github.com/gvanem/Dump1090/commit/fddf1fe74878bf308bf6cc6176e8b16cec4b4517.

Or try running with a device-arg:
`C:\Users\chad\Documents\CODE\Dump1090>dump1090.exe --interactive --net --device 0`

chadhutchins182, 2022-02-08:
That took care of it!

Thanks!!!

closed 2022-02-08 by chadhutchins182