gvanem/Dump1090 #6: Error setting frequency: -1

smholvoet, 2023-05-28:
Hi 👋

Trying to run this with my RTL-SDR Blog V3, but end with the following error when running `.\dump1090.exe`

```cmd
Found 1 device(s):
0: ??         ??                   SN: ?? (currently selected)
WARNING: Failed to enable manual gain.
WARNING: Failed to enable automatic gain.
Error setting frequency: -1.
```

It looks like it found my RTL-SDR but it fails to retrieve any info which seems odd.
I'm pretty sure my device is working properly:

[image]

[image]

Running on W11 22H2 (22621.1702).

gvanem, 2023-05-28:
I do not know what could be wrong. But in a cmd-shell (DOS-box) do:
 * `set RTLTRACE_LEVEL=1`  or `2`
 * `dump1090.exe --debug g`

And post the output.

smholvoet, 2023-05-28:
Ok something weird is going on. Somehow everything is suddenly working:

```
misc.c(377): QueryPerformanceFrequency(): 10.000 MHz
aircraft.c(745): Parsed 459998 records from: "C:/Users/Sander/repos/github/Dump1090/aircraftDatabase.csv"
Found 1 device(s):
0: Realtek    RTL2838UHIDIR        SN: 00000001 (currently selected)
Tuner gain set to automatic.
Tuned to 1090.000 MHz.
Gain reported by device: AUTO.
dump1090.c(4370): modeS_init_RTLSDR(): rc: 1.
*8d4ca813ea447865671c086d6605;
CRC: 6D6605 (ok)
RSSI: -1.2 dBFS
DF 17: ADS-B message.
  Capability     : 5 (Level 2+3+4 (DF0,4,5,11,20,21,24,code7 - is airborne))
  ICAO Address   : 4CA813 (reg-num: EI-EKT, manuf: The Boeing Company, call-sign: RYANAIR)
  Extended Squitter Type: 29
  Extended Squitter Sub : 2
  Extended Squitter Name: Unknown: 29/2
    Unrecognized ME type: 29, subtype: 2
...
```

I then tried opening SDRangel and load the ADS-B demodulator, which was unable to find the device (even though it just worked in dump1090 seconds earlier). When I then tried the above command again:

```
misc.c(377): QueryPerformanceFrequency(): 10.000 MHz
aircraft.c(745): Parsed 459998 records from: "C:/Users/Sander/repos/github/Dump1090/aircraftDatabase.csv"
Found 1 device(s):
0: ??         ??                   SN: ?? (currently selected)
WARNING: Failed to enable manual gain.
WARNING: Failed to enable automatic gain.
Error setting frequency: -1.
dump1090.c(4370): modeS_init_RTLSDR(): rc: 0.
```

Seems like it loses connection to the RTL-SDR?

gvanem, 2023-05-29:
> Seems like it loses connection to the RTL-SDR?

Try unplugging and plugging it back in. 

But I saw no `librtlsdr.c` trace in the output above. So do `set RTLTRACE_LEVEL=1` or `set RTLTRACE_LEVEL=2,winusb`
first.  I'm interested in what happens in the function `rtlsdr_set_tuner_gain_mode()`  and `r820t_set_gain_mode()`.

And the `SDR#` picture above showed you tuned to 102.1 MHz. What happens if you tune to 1090 MHz with SDR-Sharp?
Perhaps your tuner has problems with such high frequencies?

gvanem, 2023-06-06:
See this issue is solved. Hence closing. Feel free to re-open if it's not solved.

ChrisDev8, 2023-06-10:
Important to note, SDRSharp will not let other programs to access the RTL-SDR Device.

gvanem, 2023-06-10:
> SDRSharp will not let other programs access the RTL-SDR Device.

You mean: "*let other programs access it at the same time SDRSharp is using it*"?
That's since the USB-device is locked by WinUSB AFAICS.
How could sharing a USB-device like that work under the hood?