gvanem/Dump1090 #8: Update Suggestions

ChrisDev8, 2023-06-10:
## I have some suggestions for a new update of gvanem/Dump1090. 

* First, a logging upgrade with good logfile support would be perfect for large projects. Add a logging level for logfiles. Maybe a logfile for crashes or issues used for debugging purposes.

* Second, python support would be nice for automated scripts to reset Dump1090 every hour. Or to simply control Dump1090 over the network for home automation.

* Third, add support for lower sample rates to prevent the RTL-SDR device from getting very hot. I've experienced the RTL-SDR getting very hot on 2.048 M/s and especially upwards of 900 MHz.

* Fourth, add sample rate listing for easy error debugging. For example, a sample rate error would display the sample rates needed for a specific RTL-SDR device.

* Fifth, a network upgrade to allow specific ports to open instead of every port to open.

* Sixth, the gain and sample rate values should be specific to a particular RTL-SDR device. For example, the gain 40.2 may not be available on another RTL-SDR device

* Seventh, there should be no linux support because of a way better linux command line interface such as [ReADSB](https://github.com/wiedehopf/readsb). This should be on the documentation!

* Eighth, updated documentation of course duh.

gvanem, 2023-06-10:
Wow this was a lot. This should be 8 separate issues or better yet, 8 Push Requests.

> Second, python support would be nice for automated scripts to reset Dump1090 every hour. 

Why should be *reset*? Please be specific.

> Seventh, there should be no linux support because of a way better linux command line interface such as ReADSB. 
> This should be on the documentation!

Why *on the documentation*? I've built ReADSB for Windows. It's rather messy code, but the decoder works great. Mostly
due to the 2.8 MBs sample-rate.

More later.

ChrisDev8, 2023-06-10:
For the second idea, I would like python support to easily use Dump1090 with python to create some automated scripts. One example would be to reset Dump1090 every hour because sometimes it gets slow after awhile. 

For the seventh idea, people looking at the documentation should be informed about the ReADSB for linux if they have a linux computer to help anyone out.

I have no C or C++ experience so no push requests unless you or a supporter would create a push request. I rather have all eight suggestions on one issue instead of eight issues.

gvanem, 2023-07-10:
>  to reset Dump1090 every hour because sometimes it gets slow after awhile.

It's really 2 issues here; why a reset *every hour*. Why not once a day? And why / how does *it gets slow after awhile*?

ChrisDev8, 2023-07-13:
These suggestions might be trash.

closed 2023-07-13 by ChrisDev8