gvanem/Dump1090 #9: Outdated Executable & Lack of Beta Version

ChrisDev8, 2023-07-13:
# Prebuilt Executable is Too Outdated
The prebuilt executable was updated 2 months ago and needs to be refreshed. However the fix is simple. The Github Actions do not provide an Artifact for the beta version of Dump1090. You would need to enable this so that users can access the beta version of Dump1090 if they need too.

# Add Two Releases
You could also add two releases to the Github Repository. One release would be the latest Github Action Artifact and another for the latest version. This will remove the need for building the program in the main branch. The main branch will only contain the source code while the release(s) would contain two prebuilt executables for x86 and x64 architecture.

gvanem, 2023-07-13:
> You could also add two releases to the Github Repository.

How exactly? 

I regularly push an update .exe once in a while. Here is the [latest](https://github.com/gvanem/Dump1090/commit/c4b164b977440e44a02705d0a492caf5b045e692). Try it and report back please.

gvanem, 2023-07-27:
No progress. Closing.