gvanem/Dump1090 #11: Issue with wsock_trace-x64.dll not found

DrDroid, 2023-10-16:
I can run smholvoet's branch of dumnp1090 but when I run this one I get the error that "The code execution cannot proceed because wsock_trace-x64.dll was not found. Reinstalling the program may fix this problem."

I am running Windows 10 (64bit) on an AMD Ryzen 7 5800X 8-Core Processor

gvanem, 2023-10-17:
Oh, that was a slip. `wsock_trace-x64.dll` is just a helper .DLL when debugging the network traffic. 
I've now rebuilt a `dump1090.exe` w/o this .DLL. Redownload or do a `git pull` again to get it.
Or rebuilt it yourself.

DrDroid, 2023-10-17:
I appreciate the work you are doing. I can confirm the issue with wsock_trace-x64.dll is fixed. It has however revealed an underlying issue in the new way the aircraft-database.csv is handled. 

Running your compiled code it stops where there isn't an aircraft-database.csv file. When I make one using --aircrafts-update it does make a .csv and a .sqlite file but when running dump1090 again crashes at loading ...\Dump1090-main\aircraft-database.csv' could take some time.

Compiling the project in VS-Code shows an unhandled exception in the CSV_callback function Line 611 of aircraft.c

Unhandled exception at 0x00007FFE8D117FBC (ucrtbased.dll) in Dump1090.exe: An invalid parameter was passed to a function that considers invalid parameters fatal.

The issues seem to be localized to the call back as if I comment out lines 738 to 762 so the csv_callback function is never called the program seems to work fine.

I hope that helps

gvanem, 2023-10-18:
> When I make one using --aircrafts-update it does make a .csv and a .sqlite file but when running dump1090 
> again crashes at loading ...\Dump1090-main\aircraft-database.csv' could take some time.

I tried it and it indeed aborts due to Dr. Watson. I'll investigate.

gvanem, 2023-10-18:
>  I'll investigate.

Found it. It was the use of `strcpy_s()`. If you revert those changes from:
 * `strcpy_s (a, b, c);` to
 * `strncpy (a, c, b-1);` 
 
it should work.

closed 2023-10-18 by gvanem