gvanem/Dump1090 #13: Access denied: cnat get my RTL working Cygwin Win10

appaie, 2023-12-28:
Hey ive got the programm working even if im try<ing to read the testfile it works
Tried to get my data of the following RTL dongle RTL2838UHIDIR 

$ ./dump1090 --device RTL2838UHIDIR

but got following error:

Found 1 device(s):
0: ??         ??                   SN: ??
Error opening the RTLSDR device RTL2838UHIDIR: 5: Access is denied.

Following error occurs once im trying to rund dump1090 in the interactive mode

Found 1 device(s):
0: ??         ??                   SN: ?? (currently selected)
WARNING: Failed to enable manual gain.
Error setting frequency: -1.
Airports statistics:
   57421 CSV records in list.
       0 API records in list (0 dead).
       0 API requests sent. 0 received.
       0 API records cached. Used 0 times.

tried to get it working with rtl1090 without any problems see photo below:
[image]

thanks in advance für ur feedback

Cheers

gvanem, 2023-12-28:
> `5: Access is denied.`

This says it all. Some other program is using that device. Which?

appaie, 2023-12-28:
@gvanem thank u for the reply, good point i checked what could be using the rtl and saw that ive kept rtl1090 running the whole time, closed it and tried running it again it found one device but the SN: etc. are still marked with an ?

it tells me that its failed to enable manual gain, so i looked up the cfg file in which i put gain on auto

Found 1 device(s):
0: ??         ??                   SN: ?? (currently selected)
WARNING: Failed to enable manual gain.
Error setting frequency: -1.

appaie, 2023-12-28:
UPDATE: rebooted and tried it again.... it works! thanks a lot

gvanem, 2023-12-29:
> Error setting frequency: -1.

Same thing really; `Access is denied.`

>  rebooted and tried it again.... it works! 

Why? You could have unplugged the RTL-stick and put it in again. Cycling the power should completely reset it.
Besides, where does name in  `--device RTL2838UHIDIR` come from? Have you really used `rtl_eeprom.exe` to change the
name/serial-number to this incomprehensible name? Why not simply use `--device 0` (or nothing if you have only 1 stick)?

I have 2 sticks. One silver coloured and one green coloured. And have used `rtl_eeprom.exe` to give me these name:
```
c:\> rtl_eeprom -d0
Found 2 device(s):
  0:  Generic RTL2832U OEM
  1:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838-Green   << so '-d0' is this green coloured stick
Serial number:          00000000
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no
__________________________________________
```

And:
```
c:\> rtl_eeprom -d1
Found 2 device(s):
  0:  Generic RTL2832U OEM
  1:  Generic RTL2832U OEM

Using device 1: Generic RTL2832U OEM

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838-Silver    << so '-d1' is this silver coloured stick
Serial number:          00000001
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no
__________________________________________
```

Besides the colours, they are exactly the same. So using `rtl_eeprom.exe` to change their names allows me to start
Dump1090 with `--device RTL2838-green` or `--device RTL2838-silver`. No need to figure out what `0` or `1` means.

gvanem, 2024-01-03:
I guess this is fixed. Hence closing.

closed 2024-01-03 by gvanem