gvanem/Dump1090 #16: Dump1090 With RSPDuo Issues

vk6mo, 2024-02-18:
Trying to use dump1090 with RSPDuo.
Running the SDRUno software, the device receives ADS-B decodes well.

However running dump1090 with RSPDuo produces no ADS-B decodes.
Coomand line used: dump1090 --device sdrplay --interactive --net
Software starts correctly, 
1. updated databases (first time)
2. detects device 
3. correctly interrogates device (eg device serial number reported is correct)

My only unknown is which tuner it is using, so I have attached antenna to both without any successful decodes.
Statistics report 100% CRC errors. Why??

Dump file is attached (used --debug G)
[dump1090.log](https://github.com/gvanem/Dump1090/files/14321245/dump1090.log)

Running Win10 64bit with latest updates,
SDR API 3.11
Using supplied executable (dump1090.exe) from this page.
Have run from std command prompt and from administrator command prompt.

gvanem, 2024-02-18:
A SDRPlay device has never worked for me either. There's some issue with how samples are processed compared to a RTLSDR-device. 8 vs. 14 bit perl sample perhaps?

vk6mo, 2024-02-19:
OK. Thanks for getting back to me and thanks for your feedback. If I can get a windows build environment set up, I might see if I can investigate. My first thought was there was some extra config needed that wasn't set, but the samples themselves is certainly a possibility.

gvanem, 2024-03-23:
>  I might see if I can investigate. 

Any progress on this?

vk6mo, 2024-03-23:
No. I've not pursued. 
I'm using the SDRUno software itself and connecting the feeder software to that.

gvanem, 2026-01-29:
@tpglitch If this actually works, I'd be amazed.

tpglitch, 2026-01-29:
@gvanem Lol it's a simple context-based guess, we'll see. I can't test it myself because I do not own an RSPDuo.

tpglitch, 2026-01-29:
Also @gvanem what are your thoughts on adding this? Would this make it into upstream?
https://github.com/tpglitch/Dump1090/commit/e11137844cde9e71c72d1f3fa90acac29d8bd3a1

gvanem, 2026-01-31:
I do get trace like:
```
Ambiguous SDRPlay sample format detection (max16=32754 max8=128).
Ambiguous SDRPlay sample format detection (max16=25093 max8=128).
```

at every run. I do not quite follow your logic here.
But it has to to consider the value of `#define USE_8BIT_SAMPLES 0/1` in `sdrplay.c` I think.

tpglitch, 2026-01-31:
I don't really feel like working on it further at the moment, but I may if and when I get an RSPduo

gvanem, 2026-04-05:
> Do I have to rebuild with static bool use_packed_dll = true; ?

No, but what is your cmd-line?