gvanem/Dump1090 #17: Plane Jumping around on the map?

Sonic486, 2024-03-23:
now, im not sure if this is a fixable bug, or really what is going on here, but planes seem to jumping between the base station and specific points on the map, as shown below. I have no idea what's causing this to happen, but it is very... odd, and not very random either.
[image]

Sonic486, 2024-03-23:
post script note:  the longer I leave it running, the more planes accumulate in those and other areas, eventually forming a grid, pictured below.
[image]

gvanem, 2024-03-23:
I see the same myself. It's some issue with the CPR (*Compact Position Reporting*) algorithm used for message-type 17 
and sub-types  **[9..18+20,21,22]**.  Probably some math-rounding error in [`decode_CPR()`](https://github.com/gvanem/Dump1090/blob/main/src/misc.c#L2131) or a wrong usage of this function. 

I grabbed it from the [Mictronics](https://github.com/Mictronics/readsb/blob/master/track.c) repo long time ago.
So I should be improved to use this [version](https://github.com/wiedehopf/readsb/blob/dev/track.c) which I fail to understand fully. The condition for a `mm->cpr_even_valid` etc.

gvanem, 2024-03-23:
>  eventually forming a grid, pictured below.

I fail to parse this "*grid*".  Your location is not shown on the map. But I assume it's in Oklahoma?

I do not get this clustering effect. I see jumps like this just now:
[image]

when clicking on the WIF453 plane to show the tracks. Seems correct initially, but then the plane jump way up North and back to the correct position. A zoomed-up version 10 min. later:
[image]

Related to your issue?

Sonic486, 2024-03-24:
possibly, but the issues happen even without tracks shown or planes selected. I don't know anything about this, I just know that when i ran it with persist on and left it going overnight, I got the results in the second image.

gvanem, 2025-06-01:
@Sonic486 With the recent changes, this should no longer be an issue. Please D/L the latest `dump1090.exe`, try it and report if this issue is still there.

gvanem, 2025-07-06:
Closing due to lack of response. Issue certainly fixed.

closed 2025-07-06 by gvanem