gvanem/Dump1090 #19: TCP output not working on Windows 10 PC

rosen1710, 2024-10-03:
Hello,
I am currently trying to fetch data from running dump1090 on Windows 10 PC and I can't load it from _`TCP output`_ (both raw and BaseStation format) on **`localhost:30002`** and **`localhost:30003`**.

I am running the program with: **_`dump1090.exe --interactive --net`_**

I got this error message: **_`ERR_INVALID_HTTP_RESPONSE`_** (note that this is different from: **_`ERR_CONNECTION_REFUSED`_**)

[image]

I am using this version of dump1090:
C:\Users\me\Dump1090> _dump1090.exe -V_
dump1090 ver: 0.4.3 (Microsoft cl 19.40.33521, release, x64, MIMALLOC, GEN_ROUTES, NETPOLLER=select()).
Built on 7 Mar 2024.

Can someone help with this problem?

gvanem, 2024-10-03:
You cannot use port 30002 or 30003 in a browser like Chrome. Since it wants to receive traffic on port 80.
What are you trying to accomplish?

rosen1710, 2024-10-04:
I can successfully access the web server on **_`localhost:8080`_**, but I want to write my own Python program that needs to access the TCP data on port **_30002_** or **_30003_**. I have tried using Postman to retrieve data (with a **_`GET`_** request) from **_`localhost:30002`_** and **_`localhost:30003`_**, but neither works. I don't know how to load this data into my Python script. Can you tell me how to retrieve data from these ports?

gvanem, 2024-10-04:
You have to do what a `py -3 tools\SBS_client.py RAW-IN` does.

When a `dump1090.exe --net` is running, you'll get:
```
Connecting to localhost:30002
Connected to localhost:30002
*8d478ecdf8230006004ab810b46d;
*02618198228262;
*8d48ae0258c38005ac9a324d4c7b;
*8d48ae029915761a400410e42991;
*8d511066ea05f8a7013c080b707a;
*280002947313e2;
*8d47a32e580d802a14d7df346f2d;
*8d47a32e99082097b0580e28700e;
*5d48ae0213b47e;
*026181b1dcea54;
*026181b1dcea54;
*02618198228262;
*8d511066998cdf0188480df76ee2;
*a0001838ca380030a4018081f460;
*a0000298205c91b5e72820f028b7;
*8d478ecd580f104378d5cb0f2998;
*5d47a32e1be5db;
*8d47a32eea0bd8a5e53c0875ddf2;
*8d47a32ef82100000049b8f5dde8;
*a800123285e425b13000008ad819;
*8d48ae0258c3875b0e94c146452a;
*8d48ae029915761a480810c2d199;
*8d511066998cde0188480df4642c;
*a00001b1205c91b4d79820b770a1;
*a0001838ea6a1b34e02400ce7249;
*8d47a32e580d9029f2d7e170c841;^C
Received 707 bytes
```

> .. using Postman ...

No idea what that is. But it seems to ass-u-me HTTP which port 30002/30003 doesn't provide.

gvanem, 2024-10-05:
I just tried running `VirtualRadar.exe` while `dump1090.exe` was running. In Chrome it shows like this:
[image]

The VirtualRadar `Tools |  Options...` dialogue should look like this:
[image]

Note: port 30002 is used.

rosen1710, 2024-10-10:
> You have to do what a `py -3 tools\SBS_client.py RAW-IN` does.
> 
> When a `dump1090.exe --net` is running

Thank you! This is exactly what I was trying to do.

closed 2024-10-10 by rosen1710