gvanem/Dump1090 #20: Unable to download aircrafts database, link seems to be dead

nikolan123, 2024-11-23:
On first launch it tries to download https://opensky-network.org/datasets/metadata/aircraftDatabase.zip, but that link seems dead and is now a 404. The link https://opensky-network.org/datasets/metadata/ linked on the readme is also a 404
[image]
[image]

gvanem, 2024-11-24:
I also just noted the same issue. Change the link in `dump1090.cfg` to:
```diff
@@ -45,7 +45,7 @@ aircrafts = %~dp0\aircraft-database.csv
 # This can be done even when another 'dump1090.exe' is running with the same
 # `$(aircrafts).sqlite' file.
 #
-aircrafts-url = https://opensky-network.org/datasets/metadata/aircraftDatabase.zip
+aircrafts-url = https://s3.opensky-network.org/data-samples/metadata/aircraftDatabase.zip

```

nikolan123, 2024-11-25:
That fixed it, thanks

closed 2024-11-25 by nikolan123

gvanem, 2026-01-03:
> The updated URL seems to timeout.

I see the same, even no response on `ping s3.opensky-network.org`.
But that URL worked 2 days ago. Probably a temporary issue. Ref: https://downforeveryoneorjustme.com/opensky-network.org

> I switched the URL to https://registry.faa.gov/database/ReleasableAircraft.zip 

That .ZIP has a totally incompatible layout and content. Not usable.

mufaka, 2026-01-03:
I assumed it was OK because I can actually run the app and the output above stated it created 381,559 records.  With the original URL, the app fails to launch.