gvanem/Dump1090 #21: Unable to start/initialize freshly unpacked instance

kiwiscout, 2025-02-03:
My eyes lit up recently finding this repo with the ability to run tar1090 and some direct JSON queries rather than run a separate Pi
However Seem to be having difficulty with a fresh/untouched download of the compiled package on W10 (on 2 different machines)

Unfortunately --debug doesn't let me in on much other than I note if I leave the position to Windows API I see the request and then nada.

Have allowed smartscreen to launch, and even manually made a Windows Firewall rule to allow the binary incase it was something there.

The only thing I seem to get even with the default config is a message that the port includes line is not recognized. If you # that out it does the no output/no start. 
`<redacted>\Dump1090-main\dump1090.cfg(130): WARNING: Unknown key/value: 'include = ?<redacted>\Desktop\sdr-install\Dump1090-main\host-deny4.cfg', is_internal: 0.`

Only other thing I can pinpoint is if it is attempting to load databases on start that are not generated on first run? I'm seeing an error running update manually (truncated some full profile paths)

> <redacted>\sdr-install\Dump1090-main>dump1090.exe --update
Illegal parameters; db_file=sdr-install\Dump1090-main\aircraft-database.csv, url=(null).
using Sqlite file: "sdr-install\Dump1090-main\aircraft-database.csv.sqlite".
Force updating '' since it does not exist.
Force updating '' by 'py.exe -3 sdr-install\Dump1090-main\tools\gen_data.py'
/dump1090/standing-data/standing-data.zip' already exist. Assumed to be unzipped OK.
Processing aircraft_files ... wrote 1125359 bytes
Processing airport_files  ... Traceback (most recent call last):
  File "sdr-install\Dump1090-main\tools\gen_data.py", line 625, in <module>
    main()
  File "sdr-install\Dump1090-main\tools\gen_data.py", line 592, in main
    create_csv_file (airports_csv,  "airport_files",  airport_files)
  File "sdr-install\Dump1090-main\tools\gen_data.py", line 158, in create_csv_file
    append_csv_file (f, lines, None)
  File "sdr-install\Dump1090-main\tools\gen_data.py", line 148, in append_csv_file
    f.write (l)
  File "C:\Program Files\Python312\Lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'charmap' codec can't encode character '\u0132' in position 5: character maps to <undefined>

I'll welcome any suggestions to throw at it.

gvanem, 2025-02-03:
> UnicodeEncodeError: 'charmap' codec can't encode character '\u0132' in position 5: character maps to

What does `chcp` print? Try a `chcp 65001` and run `dump1090.exe --debug GN --update` as a start.
No idea what's going.

kiwiscout, 2025-02-03:
Active code page: 65001

Got me beat a bit too. I've managed to sort the Pi version to no end and have all sorts of data reads going on from it. Just not much luck in starting this :/

I did just notice an error on my part - running with the included .exe and not running make I was missing the web-pages.dll
So just amended that to directly to the tar1090 web root. But alas.

> misc.c(1033): QueryPerformanceFrequency(): 10.000 MHz
misc.c(363): rc: 2, Modes.timezone: -720 min, dst_adjust: 60 min, now: 2025/02/03, 21:08:09.740
misc.c(290): 0: 'foo'
net_io.c(1696): host_port: 'localhost:30001', name: 'localhost', addr.port: 30001
net_io.c(1708): is_ip6: 0, host: localhost, port: 30001.
net_io.c(1696): host_port: 'localhost:30002', name: 'localhost', addr.port: 30002
net_io.c(1708): is_ip6: 0, host: localhost, port: 30002.
net_io.c(1696): host_port: 'localhost:30003', name: 'localhost', addr.port: 30003
net_io.c(1708): is_ip6: 0, host: localhost, port: 30003.
dump1090.c(3342): Full-name of web_page: '\Desktop\sdr-install\Dump1090-main\web_root-Tar1090/index.html'
dump1090.c(3342): Full-name of web_page: '\Desktop\sdr-install\Dump1090-main/web-pages.dll;2'
aircraft.c(643): Aircraft Sqlite database "\Desktop\sdr-install\Dump1090-main\aircraft-database.csv.sqlite", size: 77180928.

>Update of '\temp\dump1090\aircraft-database-temp.zip' not needed before Thu Feb 13 22:08:09 2025.
aircraft.c(688): '\Desktop\sdr-install\Dump1090-main\aircraft-database.csv.sqlite' is 1 days older than the CSV-file.
using Sqlite file: "\Desktop\sdr-install\Dump1090-main\aircraft-database.csv.sqlite".
airports.c(1305): file: '\temp\dump1090\standing-data\results\aircrafts.bin', exist: 0.
Force updating '' since it does not exist.
Force updating '' by 'py.exe -3 \Desktop\sdr-install\Dump1090-main\tools\gen_data.py'
'e:/users/oblivian/temp/dump1090/standing-data/standing-data.zip' already exist. Assumed to be unzipped OK.
Processing aircraft_files ... wrote 1125359 bytes
Processing airport_files  ... Traceback (most recent call last):
  File "\Desktop\sdr-install\Dump1090-main\tools\gen_data.py", line 625, in <module>
    main()
  File "\Desktop\sdr-install\Dump1090-main\tools\gen_data.py", line 592, in main
    create_csv_file (airports_csv,  "airport_files",  airport_files)
  File "\Desktop\sdr-install\Dump1090-main\tools\gen_data.py", line 158, in create_csv_file
    append_csv_file (f, lines, None)
  File "\Desktop\sdr-install\Dump1090-main\tools\gen_data.py", line 148, in append_csv_file
    f.write (l)
  File "C:\Program Files\Python312\Lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'charmap' codec can't encode character '\u0132' in position 5: character maps to <undefined>
net_io.c(1272): Flushed 0 active connections, 0 passive, 0 unknown. Remaining bytes: 0 Rx, 0 Tx. 0 timers.

I have tried with a few additional #s to rem down to a bare minimum config without loading the airport/aircraft data for instance too.

crok, 2025-02-03:
I have the exact same issue on a Win10 machine:

```
c:\dev\Dump1090>dump1090.exe --update
RequestPermissions() failed: 0x80070005: A hozzßfÚrÚs megtagadva.
Fix your Windows settings to allow applications to access you location.
Ref: 'Start | Settings | Privacy | Location' Or do not use the '--location' option.
c:\dev\Dump1090\dump1090.cfg(130): WARNING: Unknown key/value: 'include = ?c:\dev\Dump1090\host-deny4.cfg', is_internal: 0.

Force updating 'c:\dev\Dump1090\aircraft-database.csv' since it does not exist.
Force updating 'C:\Users\rozjo\AppData\Local\Temp\dump1090\aircraft-database-temp.zip' from 'https://s3.opensky-network.org/data-samples/metadata/aircraftDatabase.zip'
Got 24163 kB.
Copied 'C:\Users\rozjo\AppData\Local\Temp\dump1090\aircraft-database-temp.csv' -> 'c:\dev\Dump1090\aircraft-database.csv'
Deleting 'c:\dev\Dump1090\aircraft-database.csv.sqlite' to force a rebuild in 'aircraft_CSV_load()'
using Sqlite file: "c:\dev\Dump1090\aircraft-database.csv.sqlite".
Loading 'c:\dev\Dump1090\aircraft-database.csv' could take some time.
Creating SQL-database 'c:\dev\Dump1090\aircraft-database.csv.sqlite'... 518999
Created 519998 records

Force updating '' since it does not exist.
Force updating '' by 'py.exe -3 c:\dev\Dump1090\tools\gen_data.py'
'C:/Users/rozjo/AppData/Local/Temp/dump1090/standing-data/standing-data.zip' already exist. Assumed to be unzipped OK.
Processing aircraft_files ... Traceback (most recent call last):
  File "c:\dev\Dump1090\tools\gen_data.py", line 625, in <module>
    main()
  File "c:\dev\Dump1090\tools\gen_data.py", line 591, in main
    create_csv_file (aircrafts_csv, "aircraft_files", aircraft_files)
  File "c:\dev\Dump1090\tools\gen_data.py", line 158, in create_csv_file
    append_csv_file (f, lines, None)
  File "c:\dev\Dump1090\tools\gen_data.py", line 148, in append_csv_file
    f.write (l)
  File "C:\Users\rozjo\AppData\Local\Programs\Python\Python312\Lib\encodings\cp1250.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'charmap' codec can't encode character '\xef' in position 63: character maps to <undefined>
```
I simply downloaded the repo as a zip and ran the update. But not matter what I do when running the dump1090.exe it just exits without any error message or anything:
```
c:\dev\Dump1090>dump1090.exe
RequestPermissions() failed: 0x80070005: A hozzßfÚrÚs megtagadva.
Fix your Windows settings to allow applications to access you location.
Ref: 'Start | Settings | Privacy | Location' Or do not use the '--location' option.
c:\dev\Dump1090\dump1090.cfg(130): WARNING: Unknown key/value: 'include = ?c:\dev\Dump1090\host-deny4.cfg', is_internal: 0.
using Sqlite file: "c:\dev\Dump1090\aircraft-database.csv.sqlite".
```
This is all I got when adding --interactive and --net
```
c:\dev\Dump1090>dump1090.exe --interactive --net
RequestPermissions() failed: 0x80070005: A hozzßfÚrÚs megtagadva.
Fix your Windows settings to allow applications to access you location.
Ref: 'Start | Settings | Privacy | Location' Or do not use the '--location' option.
c:\dev\Dump1090\dump1090.cfg(130): WARNING: Unknown key/value: 'include = ?c:\dev\Dump1090\host-deny4.cfg', is_internal: 0.
using Sqlite file: "c:\dev\Dump1090\aircraft-database.csv.sqlite".
```
It's exactly the same.

gvanem, 2025-02-03:
Regarding the `UnicodeEncodeError:` error, try a
`set PYTHONIOENCODING=:replace` before running the program.

Ref. https://docs.python.org/3/using/cmdline.html#envvar-PYTHONIOENCODING

The failure from `location_init()` (`Fix your Windows settings to allow applications to access you location.`) 
and hence failure to start is caused by `location = yes` in `dump1090.cfg`. Try `location = no` instead. 
It better default to `location = no` when I think about it.

crok, 2025-02-03:
I'm a long time user of your application (really appreciate your work!!!), just got the idea to come back and check whether there are changes and upgrade my local copy.. I did not save the working one.. and just wanted to share the fact of the same behavior (I mean: someone just downloads the zip and will probably face the very same issue). The location config is irrelevant, I did set my own location in the config (and turned off location gathering):
```
c:\dev\Dump1090>dump1090.exe --interactive --net
c:\dev\Dump1090\dump1090.cfg(130): WARNING: Unknown key/value: 'include = ?c:\dev\Dump1090\host-deny4.cfg', is_internal: 0.
using Sqlite file: "c:\dev\Dump1090\aircraft-database.csv.sqlite".
```
So I disabled to include the host-deny4.cfg in the dump1090.cfg, too, now this is the only line I got when running the exe:
```
c:\dev\Dump1090>dump1090.exe --interactive --net
using Sqlite file: "c:\dev\Dump1090\aircraft-database.csv.sqlite".
```
This is all.

gvanem, 2025-02-03:
> This is all.

Strange. If you have GNU-make, MSVC etc. try setting `USE_BIN_FILES = 0` in `Makefile.Windows` 
(or remove `-DUSE_BIN_FILES` in `Dump1090.vcxproj`) and rebuild everything:
`make -f Makefile.Windows CC=cl clean all`

kiwiscout, 2025-02-03:
It had crossed my mind if there was the potential for something recently introduced that a legacy config /install option is happy with, yet a fresh out of box one might face this.

Couldn't quite work out how to go back in time to an older commit and try short of picking a random couple of earlier ones in December. But each faced different config change hiccups.

gvanem, 2025-02-03:
The whole issue boils down to `airports_init()` returning `false` without any error-message.

kiwiscout, 2025-02-03:
I hate being 'that guy' (that normally ends up a me fault so doesn't bring them up), so glad it steered things the right direction for a potential bug/issue this time.

gvanem, 2025-02-03:
Try the ZIP now after the latest change(s): https://github.com/gvanem/Dump1090/commit/da80cdf9d131497399e3ad08c7587dee7bcd151a

crok, 2025-02-03:
> Try the ZIP now after the latest change(s): [da80cdf](https://github.com/gvanem/Dump1090/commit/da80cdf9d131497399e3ad08c7587dee7bcd151a)

Great, it runs now!

kiwiscout, 2025-02-03:
Yahoo. You complied it faster than I did. (ran out of temp space) 

Still has an issue with the host-deny include line, however that loads into the small native receive info panel like in the screenshots now. Just need some normal hours for some traffic to test.

Hopefully I can still work with the database files and customise them a bit for local traffic from my pre-populated sqlite. The appeal is the tar1090 screen and json outs for current aircraft to run some python/PHP to pull some live video overlay data.

gvanem, 2025-02-03:
> Still has an issue with the host-deny include line,

Me too. Not important. I'll maybe just remove that deny-feature.

>  to run some python/PHP to pull some live video overlay data.

Some *FlightRadar24 3D View* feature?

kiwiscout, 2025-02-03:
> > to run some python/PHP to pull some live video overlay data.
> 
> Some _FlightRadar24 3D View_ feature?

No no, just replicate basic screen ticker-tape overlays (OBS broadcast software) similar to youtube livestreams from the airport. 
Rather than do lots of direct API calls to the main tracker sites, get live data from a stick while onsite and fill in registration etc from local table

That way there is room to do some geo-distance from position calculations etc to work out what's likely in the viewfinder in the distance.

closed 2025-02-03 by kiwiscout

gvanem, 2025-02-05:
> UnicodeEncodeError: 'charmap' codec can't encode character '\xef' in position 63: character maps to <undefined>

Can some of you try this patch:
```diff
--- a/tools/gen_data.py
+++ b/tools/gen_data.py
@@ -147,6 +147,13 @@ def append_csv_file (f, lines, header):
   for l in lines:
       f.write (l)

+def append_csv_file_2 (f, lines, header):
+  if header:
+     f.write (header)
+  for l in lines:
+      _l = l.encode ("ascii", errors="backslashreplace")
+      f.write (str(_l) + "\n")
+
 def create_csv_file (to_file, name, _dict):
   f = open_file (to_file, "w+t")
   print ("Processing %-*s ... " % (len("aircraft_files"), name), end = "")
@@ -155,7 +162,7 @@ def create_csv_file (to_file, name, _dict):
       if i == 0:
          append_csv_file (f, lines, _dict[from_file]["header"])
       else:
-         append_csv_file (f, lines, None)
+         append_csv_file_2 (f, lines, None)
   f.close()
   print ("wrote %d bytes" % os.stat(to_file).st_size)
```

And rebuild with `make -f Makefile.Windows CC=cl USE_BIN_FILES=1 clean all`.

gvanem, 2025-02-05:
That was wrong. Try:
```diff
--- a/tools/gen_data.py
+++ b/tools/gen_data.py
@@ -141,21 +141,26 @@ def read_csv_file (fname, _dict):
   f.close()
   return lines

-def append_csv_file (f, lines, header):
+def append_csv_file (f, lines, header, ignore):
   if header:
      f.write (header)
+
   for l in lines:
-      f.write (l)
+      if ignore:
+         _l = l.encode ("ascii", errors="backslashreplace")
+         f.write (str(_l) + "\n")
+      else:
+         f.write (l)

-def create_csv_file (to_file, name, _dict):
+def create_csv_file (to_file, name, _dict, ignore):
   f = open_file (to_file, "w+t")
   print ("Processing %-*s ... " % (len("aircraft_files"), name), end = "")
   for i, from_file in enumerate(_dict):
       lines = read_csv_file (from_file, _dict)
       if i == 0:
-         append_csv_file (f, lines, _dict[from_file]["header"])
+         append_csv_file (f, lines, _dict[from_file]["header"], False)
       else:
-         append_csv_file (f, lines, None)
+         append_csv_file (f, lines, None, ignore)
   f.close()
   print ("wrote %d bytes" % os.stat(to_file).st_size)

@@ -588,9 +593,9 @@ def main():
      print ("total_fsize: %s" % nice_size(total_fsize))
      sys.exit (0)

-  create_csv_file (aircrafts_csv, "aircraft_files", aircraft_files)
-  create_csv_file (airports_csv,  "airport_files",  airport_files)
-  create_csv_file (routes_csv,    "route_files",    route_files)
+  create_csv_file (aircrafts_csv, "aircraft_files", aircraft_files, True)
+  create_csv_file (airports_csv,  "airport_files",  airport_files, True)
+  create_csv_file (routes_csv,    "route_files",    route_files, False)

   num_aircrafts = create_bin_file (aircrafts_bin, aircrafts_csv, len(aircraft_files), aircraft_rec_len, aircraft_record)
   num_airports  = create_bin_file (airports_bin,  airports_csv,  len(airport_files),  airport_rec_len,  airport_record)
```

gvanem, 2025-02-05:
Forget the above; this Unicode stuff in Python drives me crazy. Seems I have to create `airports.csv` as a binary file.