gvanem/Dump1090 #22: Invalid epos (Distance caculator from homepos?)

kiwiscout, 2025-02-04:
In the logs I'm seeing the following being produced. With no correct distance calc - related to me being in the southern hemisphere or an offset calculation for latitude?

> 20:52:39.177: addr C81AC0: Invalid epos: +172.710 lon, +133.144 lat from heading:  -138.0. delta_X: -164.570, delta_Y: -182.773.
> 20:52:39.458: addr C81AC0: Invalid epos: +172.708 lon, +133.145 lat from heading:  -138.0. delta_X:  -54.724, delta_Y:  -60.778.
> 20:52:40.586: addr C81AC0: Invalid epos: +172.708 lon, +133.145 lat from heading:  -138.0. delta_X: -111.432, delta_Y: -123.757.
> 20:52:40.866: addr C81AC0: Invalid epos: +172.706 lon, +133.146 lat from heading:  -138.0. delta_X:  -55.121, delta_Y:  -61.218.

At first I thought it was my lat/long settings so triple checked those (I think the readme potentially has them backwards in the example?)

> If a config-setting homepos = longitude,latitude setting is defined, the distance to the place gets calculated. I.e. the Dist column above. E.g. a homepos = 60.3016821,5.3208769 for Bergen/Norway.

Which is lat 60 long 8?

gvanem, 2025-02-04:
> If a config-setting homepos = longitude,latitude setting is defined, 

Yes, that's backward. Should be `homepos = latitude,longitude` ...
(the name *longitude* comes from the length of the Mediterranean Sea).

I also have some problems with the decoding of the CPR in some cases; I assume that's where you're problem also comes from.

I replayed this flight: https://planefinder.net/data/flight/NZ5798/history/5-5651005 to get a clue.

There could be a problem with a too simple `cartesian_to_spherical()` and the use of `atan2()` 
in this case. You're in New-Zealand (no?).  So the estimated position should be close to **+172.710 lon, -46.856 lat**.
Somewhere there is a missing mapping to **[-90, 90]** for latitude.

gvanem, 2025-05-25:
I have a long list of commits that now seems to have fixed this.
The CPR stuff is completely rewritten thanks to Dump1090-FA and Dump1090-RSP.
Stay tuned for those `git push` updates.

kiwiscout, 2025-05-29:
If I can grab some more data or test something let me know if one of those recent ones was meant to help.

As unfortunately.. 
00:23:58.804: plane C82759: Invalid epos: +172.515 lon, +133.069 lat from heading:  +100.0. delta_X: +110.110, delta_Y:  -19.415.

gvanem, 2025-06-01:
I'm disabling the call to `aircraft_set_est_home_distance()` until I figure out what's wrong.
So this issue is no longer relevant in the next commit.