gvanem/Dump1090 #23: HTTP display fields empty

kiwiscout, 2025-05-25:
Bit of an oddity in Web displays I'm seeing I wouldn't mind getting confirmed by others.

(Someone mentioned it online and have just confirmed)

Tar1090 map appears to not be able to grab the Sqawks from decodes
While the FlightAware map is unable to display Squawk, Altitude and Velocity/Speed.

Am seeing them in DF17/20s from the console while it is decoding. And interactive. Just not transposed to the web sources.

kiwiscout, 2025-05-25:
> Reference?

User here. https://discussions.flightaware.com/t/updated-piaware-skyview-no-altitude-or-speed-display/41895/4

Was me that replied to him to say check in. But gathered he may not be able to describe or test it as thoroughly so tested myself  when I was able.

> You mean this field in Tar1090 looks like:
> [image]

Not quite. The tracked aircraft data tables. But that does translate down to N/A for the fields when selecting a target.

[image]

> Some JSON-output is missing. I think it expects `"Sqk:xx"` from `a->squawk`, but it's not sent. Can you check with Chrome Dev-console (if you have it) to see what Dump1090 sends?
> 
> But a **huge** number of commit are coming up which could have fixed this.

Not a lot in Console. Some other other core jsons piaware recently added to onloads. It is expecting status.json (the piaware receiver/server state) and upintheair.json. But shouldn't be related.

Actually I just noticed the aicraft.json generated/called in request/response is indeed not parsing a sqwk or similar. But it does have altitude and speed I would expect to populate.

{"now":1748165929.052, "messages":681, "aircraft":
[{"hex": "c82336", "flight": "ANZ329M", "lat": -43.532547, "lon": 172.481583, "altitude": 1050, "track": 41, "speed": 94, "type": "adsb_icao", "messages": 375, "seen": 2, "seen_pos": 1, "rssi": -49.5},
{"hex": "c82871", "flight": "ANZ390M", "lat": -42.444096, "lon": 173.682031, "altitude": 19000, "track": 41, "speed": 247, "type": "adsb_icao", "messages": 306, "seen": 2, "seen_pos": 1, "rssi": -49.5}]
}

gvanem, 2025-05-25:
>  from a->squawk, but it's not sent.

Correction That should be  `a->identity`.

gvanem, 2025-05-25:
> Actually I just noticed the aicraft.json generated/called in request/response is indeed not parsing a sqwk or similar. 

I know. But it an easy fix soon to be commited.

kiwiscout, 2025-05-29:
OK so the last compiled .exe included appears somewhat broken. For me at least.

I managed to successfully compile myself from the latest code (had to edit the vcxproj to output /release/dump1090.exe not ../dump1090.exe to prevent errors) and it's now not force closing.

FYI - noted there's also duplicate SDR GAINs set in the default config file. Auto and 31

Back to this version, Tar1090 html squawk field is populated. As it is the SkyAware map now too. 

However SkyAware missing populating from the /data/aircraft.json:
Registration, Type, Altitude, Speed, Vert Rate

And presumably also not hot-querying the sqlite/csv for the AC detail

[image]

Appreciate your continued for this BTW. Can see it making lives easier for the not so Raspberry Pi inclined. 
I know I could make double use of it for a specific use case with extended readsb-like JSON reads .  But I don't think those are tar1090 native. https://github.com/wiedehopf/readsb/blob/dev/README-json.md

kiwiscout, 2025-05-29:
I think this is a diff...

When I compiled using msbuild there was no icon on the .exe. And it tried to merge it to the existing when launched from the src folder. So created appicon.rc and adjusted where to output the final .exe

Thanks Dr google.

```
@@ -18,6 +18,9 @@
       <Platform>x64</Platform>
     </ProjectConfiguration>
   </ItemGroup>
+  <ItemGroup>
+  <ResourceCompile Include="appicon.rc" />
+</ItemGroup>
   <PropertyGroup Label="Globals">
     <VCProjectVersion>16.0</VCProjectVersion>
     <ProjectGuid>{E4F9EDE0-9C94-4F7F-9E30-6B9BB0948E60}</ProjectGuid>
@@ -88,7 +91,7 @@
     <Link>
       <TargetMachine>MachineX86</TargetMachine>
       <GenerateDebugInformation>true</GenerateDebugInformation>
-      <OutputFile>../dump1090.exe</OutputFile>
+      <OutputFile>/release/dump1090.exe</OutputFile>
       <SubSystem>Console</SubSystem>
       <AdditionalDependencies>advapi32.lib;dnsapi.lib;iphlpapi.lib;ole32.lib;user32.lib;setupapi.lib;winusb.lib;ws2_32.lib</AdditionalDependencies>
       <AdditionalOptions>/IGNORE:4099</AdditionalOptions>
@@ -105,7 +108,7 @@
     <Link>
       <TargetMachine>MachineX86</TargetMachine>
       <GenerateDebugInformation>true</GenerateDebugInformation>
-      <OutputFile>../dump1090.exe</OutputFile>
+      <OutputFile>release/dump1090.exe</OutputFile>
       <SubSystem>Console</SubSystem>
       <OptimizeReferences>false</OptimizeReferences>
       <AdditionalDependencies>advapi32.lib;dnsapi.lib;iphlpapi.lib;ole32.lib;user32.lib;setupapi.lib;winusb.lib;;ws2_32.lib</AdditionalDependencies>

```

kiwiscout, 2025-05-29:
> That should be `<ResourceCompile Include="dump1090.rc" />` since it's already there with an .ICO-file inside.

Ah. I blindly made a fresh one after asking AI.

> Then one has to manually copy/install it to `..`.
> 
> And why does MSbuild bitches on `../dump1090.exe`?

J:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(139
2,5): warning MSB8012: TargetPath(E:\Users\\Desktop\sdr-install\Dump1090-main\src\Release\Dump1090.exe) does no
t match the Linker's OutputFile property value (E:\Users\\Desktop\sdr-install\Dump1090-main\dump1090.exe). This
 may cause your project to build incorrectly. To correct this, please make sure that $(OutDir), $(TargetName) and $(Tar
getExt) property values match the value specified in %(Link.OutputFile). [E:\Users\\Desktop\sdr-install\Dump109
0-main\src\Dump1090.vcxproj]

It did something strange yesterday after getting that. It would run, but on quit was hidden from the file system and caused write failures trying to replace it from the archive again until restart. 

So to avoid that tonight made the expected output locations match (no more errors) and just copied it over