gvanem/Dump1090 #24: Interactive Web server broken?

Sonic486, 2025-06-07:
The program creates the web server properly, however, the web browser does not connect. 

I ran the program with the included file run.py file, which started it with the interactive web server 
( D:\Dump1090-main\dump1090.exe --net  --interactive --device 0 )

I ran netstat, and it shows port 8080 listening on the localhost 

  Proto  Local Address          Foreign Address        State           PID
  TCP    0.0.0.0:8080           0.0.0.0:0              LISTENING       431728

I then ran tasklist and it shows PID 431728 as dump1090.exe, so this confirms and open TCP web socket 

  Image Name                         PID Session Name        Session#    Mem Usage
  =========== ======== ================ =========== 
  dump1090.exe                431728 Console                    1     42,084 K

However, attempting to open 0.0.0.0:8080, 127.0.0.1:8080, in any web browser on same machine does no open a web page, instead returning 

  ERR_ADDRESS_INVALID 

or

 ERR_CONNECTION_REFUSED

respectively.

what should I do from here to help diagnose/resolve this issue?

gvanem, 2025-06-07:
> The program creates the web server properly, however, the web browser does not connect.

I just D/L the .ZIP of this project and ran `dump1090.exe --net --interactive --device 0`.
It works fine. 

Did you allow it in the *Windows Firewall* dialogue? Or it could be a Web-browser cache issue.
Try pressing **Ctrl-F5** (or **Shift-F5**) to clear the web-cache for the Dump1090 web-server first.
Ref. https://www.filecloud.com/blog/2015/03/tech-tip-how-to-do-hard-refresh-in-browsers/

gvanem, 2025-06-07:
> ERR_ADDRESS_INVALID

Which address and who wrote that?

> ERR_CONNECTION_REFUSED

Ditto. It could be you have an issue with IPv6. Try `http-ipv6 = true` in `dump1090.cfg`.

gvanem, 2025-07-06:
No response; seems fixed. Closing.

closed 2025-07-06 by gvanem