gvanem/Dump1090 #25: data/aircraft.json does not return aircraft

tpglitch, 2025-07-04:
data/aircraft.json does not return aircraft

What I did:
- Downloaded Dump1090
- Launched it, on Windows, with flags net and interactive

The interactive terminal does show aircraft:

[image]

And using Virtual Radar Server, connected on port 30002

[image]
we have data...

But tar1090 is different. Going to port 8080 brings up tar1090 but shows nothing

[image]

I investigated and found this return at /data/aircraft.json:

[image]

Help is appreciated!

gvanem, 2025-07-04:
Strange. I just D/L the .ZIP from here and ran `dump1090.exe --net --interactive`.

The Debug-console in Chrome shows a lot of data in `aircraft.json`:

[image]

tpglitch, 2025-07-04:
Very interesting. As you are the maintainer, what steps would you suggest I take to debug the issue. I am using the RTL-SDR Blog V4, on Windows 10 22H2 Build 19045.5965. Tried Chromium and Firefox just in case, and I'm using WinUSB v6.1.7600.16385.

gvanem, 2025-07-04:
Terminal only. No network unless you say `dump1090.exe --raw --net`. 

Option `--raw` shows data on raw form like this:
```
*8d479227ea05f855e53c08858079;
*2000023d257243;
*5d47922734f037;
*8dc079d5990d4028c00418033610;
...
```

tpglitch, 2025-07-04:
Unable to build..
```
  Dump1090.vcxproj -> C:\Users\tpglitch\Desktop\Dump1090\src\Release\Dump1090.exe
FinalizeBuildStatus:
  Deleting file "Dump1090\Release\Dump1090.tlog\unsuccessfulbuild".
  Touching "Dump1090\Release\Dump1090.tlog\Dump1090.lastbuildstate".
Done Building Project "C:\Users\tpglitch\Desktop\Dump1090\src\Dump1090.vcxproj" (default targets).

Done Building Project "C:\Users\tpglitch\Desktop\Dump1090\src\Dump1090.sln" (default targets).

Build succeeded.

"C:\Users\tpglitch\Desktop\Dump1090\src\Dump1090.sln" (default target) (1) ->
"C:\Users\tpglitch\Desktop\Dump1090\src\Dump1090.vcxproj" (default target) (2) ->
(ClCompile target) ->
  C:\Users\tpglitch\Desktop\Dump1090\src\interactive.c(275,31): warning C4018: '<=': signed/unsigned mismatch [C:\Users
\tpglitch\Desktop\Dump1090\src\Dump1090.vcxproj]

"C:\Users\tpglitch\Desktop\Dump1090\src\Dump1090.sln" (default target) (1) ->
"C:\Users\tpglitch\Desktop\Dump1090\src\Dump1090.vcxproj" (default target) (2) ->
(DoLinkOutputFilesMatch target) ->
  C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(1402,5):
 warning MSB8012: TargetPath(C:\Users\tpglitch\Desktop\Dump1090\src\Release\Dump1090.exe) does not match the Linker's O
utputFile property value (C:\Users\tpglitch\Desktop\Dump1090\dump1090.exe). This may cause your project to build incorr
ectly. To correct this, please make sure that $(OutDir), $(TargetName) and $(TargetExt) property values match the value
 specified in %(Link.OutputFile). [C:\Users\tpglitch\Desktop\Dump1090\src\Dump1090.vcxproj]
```

tpglitch, 2025-07-04:
Yeah i realized that after looking thru the configs

tpglitch, 2025-07-04:
> It works w/o any network? Like dump1090.exe --raw.
Confirmed! Packets are received

[image]

Still no aircraft, however :(

[image]

tpglitch, 2025-07-04:
I also tried earlier versions, still nothing. Something about my Windows environment has to be different that the program is just bugging out, but I don't know what it would be.

gvanem, 2025-07-04:
> Something about my Windows environment has to be different that the program is just bugging out, 

Attach the `dump1090.log` file (as `dump1090.log.txt`)

tpglitch, 2025-07-04:
Errors begin to show up due to `jq.exe`'s absence.
```

aircraft_test_3(): Generate and check JSON-data:
  Dumping 50 aircrafts (8568 bytes) to 'C:\Users\tpglitch\AppData\Local\Temp\dump1090\json-1.txt'
'jq.exe' is not recognized as an internal or external command,
operable program or batch file.
  Command 'jq.exe < C:\Users\tpglitch\AppData\Local\Temp\dump1090\json-1.txt > NUL' failed; rc: 1.

  Dumping 50 aircrafts (10669 bytes) to 'C:\Users\tpglitch\AppData\Local\Temp\dump1090\json-2.txt'
'jq.exe' is not recognized as an internal or external command,
operable program or batch file.
  Command 'jq.exe < C:\Users\tpglitch\AppData\Local\Temp\dump1090\json-2.txt > NUL' failed; rc: 1.

  Dumping 0 aircrafts (2 bytes) to 'C:\Users\tpglitch\AppData\Local\Temp\dump1090\json-3.txt'
'jq.exe' is not recognized as an internal or external command,
operable program or batch file.
  Command 'jq.exe < C:\Users\tpglitch\AppData\Local\Temp\dump1090\json-3.txt > NUL' failed; rc: 1.

  Dumping 0 aircrafts (54 bytes) to 'C:\Users\tpglitch\AppData\Local\Temp\dump1090\json-4.txt'
'jq.exe' is not recognized as an internal or external command,
operable program or batch file.
  Command 'jq.exe < C:\Users\tpglitch\AppData\Local\Temp\dump1090\json-4.txt > NUL' failed; rc: 1.

C:\Users\tpglitch\Desktop\Dump1090\src\airports.c(1259, main-thread): c_in: 0xFFFFFFFF, state: 6
Airports statistics:
   57421 CSV records in list.
       0 API records in list (0 dead).
       0 API requests sent. 0 received (HTTP 404: 0)
       0 API records cached. Used 0 times.
```

Reviewing the files shows they were not created.

[image]

tpglitch, 2025-07-04:
I will now download `jq.exe`, and place it in a directory included in PATH. I will get back soon.

tpglitch, 2025-07-04:
Aircrafts do show up in two of the JSON test. Here's two:

[image]

[image]

As you can see, test 3 and 4 have 0 aircrafts.

[image]

gvanem, 2025-07-04:
> I will now download jq.exe

Does not matter for the generation of `%TEMP%\dump1090\json-*.txt`. 

I use `jq.exe` only to check the validity of those TEMP-files.

tpglitch, 2025-07-04:
Ah, I must have misread the error. I have uploaded test results above

gvanem, 2025-07-04:
The picture above says `Image rejection not calibrated yet`. Never seen this, but you could try with
`rtlsdr-calibrate = yes` in `dump1090.cfg` are retry.

gvanem, 2025-07-04:
> PS. If you have your location enabled in Windows,

I think that's the problem with your setup; the `homepos` is way wrong. Hence the calculated CPR-relative position will be
totally wrong. Zoom out the map in Tar1090 and you should see some planes.

tpglitch, 2025-07-04:
Could add a setup script which runs a TUI wizard asking to setup location?

gvanem, 2025-07-04:
>  wizard asking to setup location?

Hardly anybody know their exact position w/o going to GoogleMap or https://www.freemaptools.com/elevation-finder.htm

tpglitch, 2025-07-04:
I'm working on a setup script test right now to see if and how it would work

gvanem, 2025-07-05:
I'm closing this now. But I'd like to detect such problems at runtime with some heuristics. Not sure what yet.

closed 2025-07-05 by gvanem